asus-linux-drivers/asus-dialpad-driver #2: [Bug]: Asus Proart P16 2024 (H7606W) error

GLLM, 2025-02-26:
### Describe the bug

Hi @ldrahnik 

Thanks for all the work on the dialpad!

I installed the driver (both automatically and manually).
During install I see this error:
```
to be added
 in a minute
```
Then I reboot.
The dialpad cannot be activated (swipe or long press in the right corner). The LED never switches On.

Not sure why that happens, since the python draft on dialpad was working nicely.

Thanks
GLLM

### Expected behavior

dialpad can be activated, used and deactivated

### Relevant log output

```shell
coming up
```

### Desktop

Fedora 41 KDE spin + NVidia 570 (wayland)
ROG kernel 6.13.3
more later

ldrahnik, 2025-02-26:
@GLLM Was missing `pydbus` dependency used on `wayland` (pushed now). Could help.

GLLM, 2025-02-26:
Pulled, uninstalled, reinstalled  .... to no avail

systemctl status --user asus_dialpad_driver@$USER.service:
```
$ systemctl status --user asus_dialpad_driver@$USER.service
× asus_dialpad_driver@gllm.service - Asus DialPad Driver
     Loaded: loaded (/usr/lib/systemd/user/asus_dialpad_driver@.service; enabled; preset: disabled)
    Drop-In: /usr/lib/systemd/user/service.d
             └─10-timeout-abort.conf
     Active: failed (Result: exit-code) since Wed 2025-02-26 19:55:13 CET; 23s ago
   Duration: 670ms
 Invocation: 8bf0aed9e3af4e4287518772850dd948
    Process: 6411 ExecStart=/usr/share/asus-dialpad-driver/.env/bin/python3 /usr/share/asus-dialpad-driver/dialpad.py proartp16 /usr/share/asus-dialpad-driver/ (co>
   Main PID: 6411 (code=exited, status=1/FAILURE)

févr. 26 19:55:13 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Scheduled restart job, restart counter is at 21.
févr. 26 19:55:13 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Start request repeated too quickly.
févr. 26 19:55:13 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Failed with result 'exit-code'.
févr. 26 19:55:13 p16 systemd[2392]: Failed to start asus_dialpad_driver@gllm.service - Asus DialPad Driver.
```

journalctl -f --user-unit asus_dialpad_driver@$USER.service: 
```
févr. 26 19:55:10 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Main process exited, code=exited, status=1/FAILURE
févr. 26 19:55:10 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Failed with result 'exit-code'.
févr. 26 19:55:11 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Scheduled restart job, restart counter is at 20.
févr. 26 19:55:11 p16 systemd[2392]: Started asus_dialpad_driver@gllm.service - Asus DialPad Driver.
févr. 26 19:55:12 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Main process exited, code=exited, status=1/FAILURE
févr. 26 19:55:12 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Failed with result 'exit-code'.
févr. 26 19:55:13 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Scheduled restart job, restart counter is at 21.
févr. 26 19:55:13 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Start request repeated too quickly.
févr. 26 19:55:13 p16 systemd[2392]: asus_dialpad_driver@gllm.service: Failed with result 'exit-code'.
févr. 26 19:55:13 p16 systemd[2392]: Failed to start asus_dialpad_driver@gllm.service - Asus DialPad Driver.
```

ldrahnik, 2025-02-26:
@GLLM And when you run manually ditectly the script like previously in the NumberPad driver issue? `dialpad.py`

GLLM, 2025-02-26:
@ldrahnik 

python dialpad.py (does not run, whereas it did when I was using the numberpad install+manual dialpad.py run):
```
$ sudo py dialpad.py 
2025-02-26 22:02:38,393 ERROR Failed to connect to Wayland display: Unable to connect to display
```

/usr/share/asus-dialpad-driver/.env/bin/python3 /usr/share/asus-dialpad-driver/dialpad.py proartp16 /usr/share/asus-dialpad-driver/:
```
$ /usr/share/asus-dialpad-driver/.env/bin/python3 /usr/share/asus-dialpad-driver/dialpad.py proartp16 /usr/share/asus-dialpad-driver/
2025-02-26 22:03:16,073 INFO Wayland session detected and connected.
2025-02-26 22:03:16,074 INFO Detecting touchpad from string: "N: Name="ASCF1A01:00 2808:0234 Touchpad""
2025-02-26 22:03:16,074 INFO Set touchpad device id 0 from S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASCF1A01:00/0018:2808:0234.0004/input/input18
2025-02-26 22:03:16,074 INFO Set touchpad id 8 from H: Handlers=mouse3 event8
2025-02-26 22:03:16,074 INFO Touchpad min-max: x 0-4762, y 0-3099
2025-02-26 22:03:16,075 INFO Virtual device initialized successfully.
2025-02-26 22:03:16,576 ERROR Listening touchpad events unexpectedly failed
Traceback (most recent call last):
  File "/usr/share/asus-dialpad-driver/dialpad.py", line 698, in <module>
    load_all_config_values()
    ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/share/asus-dialpad-driver/dialpad.py", line 295, in load_all_config_values
    suppress_app_specifics_shortcuts = int(config_get(CONFIG_SUPPRESS_APP_SPECIFICS_SHORTCUTS, CONFIG_SUPPRESS_APP_SPECIFICS_SHORTCUTS_DEFAULT))
ValueError: invalid literal for int() with base 10: 'config_supress_app_specifics_shortcuts_default'
2025-02-26 22:03:16,577 INFO Clean up started
2025-02-26 22:03:16,577 INFO Clean up finished
2025-02-26 22:03:16,577 INFO Exiting
```

ldrahnik, 2025-02-27:
@GLLM Try again `/usr/share/asus-dialpad-driver/.env/bin/python3 /usr/share/asus-dialpad-driver/dialpad.py proartp16 /usr/share/asus-dialpad-driver/`

GLLM, 2025-02-28:
@ldrahnik i pulled the latest commit and reinstalled
It works nicely now ! <3

I think i can close this issue !

I have a few suggestions, Can to hear about those ? in new issues ("new feature" type) ?

thanks
GLLM

ldrahnik, 2025-02-28:
Yes, please open new issues.

closed 2025-02-28 by ldrahnik