asus-linux-drivers/asus-dialpad-driver #6: [Feature Request] Hide mouse cursor when scrolling through dialpad

GLLM, 2025-02-28:
### Short Description

Hi @ldrahnik 

when I scroll the Dialpad, I can see the mouse cursor making happy round tour on my screen 😆 

I believe mouse cursor could be hidden when using the dialpad since the active functionality this is defined to trigger most often will display the effect (OSD display or differently).

Do you think this is something feasible ?
Many thanks
GLLM

ldrahnik, 2025-03-01:
@GLLM Should be fixed in the latest master.

GLLM, 2025-03-01:
Hello @ldrahnik 
I've pulled, updated but sadly, it does not work.
See my screen recording: 

https://github.com/user-attachments/assets/2fd97e25-8a7b-454d-92b0-5e3f8e20f928

Thanks
GLLM

ldrahnik, 2025-03-01:
@GLLM Does enable/disable touchpad `qdbus` command? `qdbus org.kde.KWin '/org/kde/KWin/InputDevice/event/XY' org.kde.KWin.InputDevice.sendEvents '1|0'`, 0/1 at the end for disabling and enabling and change the event number to appropriate for the touchpad

ldrahnik, 2025-03-01:
@GLLM This command should list your devices `qdbus org.kde.KWin /org/kde/KWin/InputDevice`

ldrahnik, 2025-03-02:
@GLLM Or try for disabling touchpad:

```
qdbus org.freedesktop.DeviceKit.Hal /org/freedesktop/DeviceKit/Hal/Manager org.freedesktop.DeviceKit.Hal.Device.SetProperty 'usb_device' 'touchpad_device' 'false'
```

and for enabling touchpad:

```
qdbus org.freedesktop.DeviceKit.Hal /org/freedesktop/DeviceKit/Hal/Manager org.freedesktop.DeviceKit.Hal.Device.SetProperty 'usb_device' 'touchpad_device' 'true'```

GLLM, 2025-03-02:
I'm having a hard time getting that info.
I'm perusing the internetz to try get ahold of that info ... to no avail so far.

ldrahnik, 2025-03-13:
@GLLM What about this bash script:

```
#!/bin/bash

# Duration in seconds (change this as needed)
DURATION=3

# Find touchpad device ID
DEVICE_ID=$(libinput list-devices | grep -i -B 1 'touchpad' | grep 'Device:' | awk -F 'Device: ' '{print $2}')

# Disable touchpad
echo "Disabling touchpad: $DEVICE_ID"

qdbus org.kde.KWin /org/kde/KWin/InputDevice/touchpad org.kde.KWin.InputDevice.setEnabled false

# Wait for the specified duration
sleep $DURATION

# Enable touchpad
echo "Re-enabling touchpad: $DEVICE_ID"

qdbus org.kde.KWin /org/kde/KWin/InputDevice/touchpad org.kde.KWin.InputDevice.setEnabled true
```

Also `qdbus` but with different path, looks like this is more related to `KWin` that you use.

GLLM, 2025-03-13:
@ldrahnik thanks !

with the webpage, I found it : 
```
gllm@p16:~/Downloads$ grep touchpad /sys/class/input/event*/device/name -i
/sys/class/input/event9/device/name:ASCF1A01:00 2808:0234 Touchpad

```
yet, here's the result:
```
gllm@p16:~/Downloads$ qdbus org.kde.KWin '/org/kde/KWin/InputDevice/event9' org.kde.KWin.InputDevice.setEnabled true
Cannot find 'org.kde.KWin.InputDevice.setEnabled' in object /org/kde/KWin/InputDevice/event9 at org.kde.KWin

```

I'm not sure what is wrong!

Here are my setup info, if it can help.

Operating System: Fedora Linux 41
KDE Plasma Version: 6.3.2
KDE Frameworks Version: 6.11.0
Qt Version: 6.8.2
Kernel Version: 6.13.4-202.rog.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 24 × AMD Ryzen AI 9 HX 370 w/ Radeon 890M
Memory: 61.9 GiB of RAM
Graphics Processor: AMD Radeon Graphics
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: ProArt P16 H7606WI_H7606WI
System Version: 1.0

ldrahnik, 2025-03-14:
@GLLM And what about this? It works in Virtualbox. 

```
Operating System: Fedora Linux 41
KDE Plasma Version: 6.2.1
KDE Frameworks Version: 6.7.0
Qt Version: 6.7.2
Kernel Version: 6.13.6-200.fc41.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 4 × 11th Gen Intel® Core™ i5-1135G7 @ 2.40GHz
Memory: 5.7 GiB of RAM
Graphics Processor: llvmpipe
Manufacturer: innotek GmbH
Product Name: VirtualBox
System Version: 1.2
 
ldrahnik@fedora:~$ sudo libinput debug-events
-event0   DEVICE_ADDED            Power Button                      seat0 default group1  cap:k
-event1   DEVICE_ADDED            Sleep Button                      seat0 default group2  cap:k
-event5   DEVICE_ADDED            VirtualBox mouse integration      seat0 default group3  cap:p left scroll-nat calib
-event4   DEVICE_ADDED            VirtualBox USB Tablet             seat0 default group4  cap:p left scroll-nat calib
-event2   DEVICE_ADDED            AT Translated Set 2 keyboard      seat0 default group5  cap:k
-event3   DEVICE_ADDED            ImExPS/2 Generic Explorer Mouse   seat0 default group6  cap:p left scroll-nat scroll-button
-event2   KEYBOARD_KEY            +0.000s       *** (-1) pressed
 event2   KEYBOARD_KEY            +0.115s       *** (-1) pressed
^C
ldrahnik@fedora:~$ ^C
ldrahnik@fedora:~$ ^C
ldrahnik@fedora:~$ qdbus org.kde.KWin /org/kde/KWin/InputDevice/event4 org.freedesktop.DBus.Properties.Set org.kde.KWin.InputDevice "enabled" "false"
 
ldrahnik@fedora:~$ qdbus org.kde.KWin /org/kde/KWin/InputDevice/event4 org.freedesktop.DBus.Properties.Set org.kde.KWin.InputDevice "enabled" "true"
 
ldrahnik@fedora:~$ 
```

ldrahnik, 2025-03-17:
Not replicated. Can be reopened.

GLLM, 2025-03-17:
Hello @ldrahnik 
Your last message helped.
I did see I need to action the event11
`gllm@p16:~$ grep touchpad /sys/class/input/event*/device/name -i/sys/class/input/event11/device/name:ASCF1A01:00 2808:0234 Touchpad
`

I'm not sure why, but in my last message, it was on event9 !
But once using the 2 code lines, it works as expected : the mouse cursor is hidden.

However, the asus-dialpad does not work in the sense the cursor is still visible.
How can this be integrated ?
Thanks
GLLM

ldrahnik, 2025-03-18:
@GLLM Sorry. I missed that `sendEvents` was implemented instead of `enabled`. Try the latest master branch now.

GLLM, 2025-03-20:
I updated and reinstalled, but I still see the cursor when using the dialpad.
Am I expecting something I should not ?
Thanks
GLLM

ldrahnik, 2025-03-20:
@GLLM Now it has to work. Tested on Fedora in VirtualBox. Sorry again.

GLLM, 2025-03-21:
Don't be sorry !! 😃 

I've tried it. It works the first few times, and then stopped working.
I need to check how I can reproduce consistently. I'll let you know.
Thanks a lot

ldrahnik, 2025-03-21:
@GLLM And now? The latest master.

GLLM, 2025-03-22:
You did it !
It works 🥳 

Tested on Fedora 41 KDE w/ kernel 6.13.7, in different apps (Freetube, QmmP, Firefox).

**Thanks a lot !**

Now ... I'm wondering if the brightness of the center LED can be dimmed down ... as it is very bright  .... talking about never satisfied users ;)

GLLM

ldrahnik, 2025-03-22:
@GLLM Good :) Try uninstall this driver for DialPad, install NumberPad driver and let me know whether after activating via top right icon holding top left corner spot (not existing painted icon in your Touchpad there but it does not matter) change the brightness or not.