asus-linux-drivers/asus-dialpad-driver #8: [Feature Request] Ability to dim light on dialpad center led (ProArt P16 2024, aka H7606W)

GLLM, 2025-03-24:
### Short Description

Hello @ldrahnik 
As discussed on another issue, the LED in the center of the dialpad, when activated, on ProArt P16 2024, is very bright and a bit bothering (even in medium-light environment).

Could we try to dim the light ?

I've installed asus-number-pad to try (after unsintalling asus-dial-pad), but the top-right button does not do anything anymore !!
I've also tried the ./tests/test_brightness.py ... It lighted the LED but I do not see the change on each step.
I've tried to press the top-left corner but, damn, I would not be able to see if my eyes play me trick after this long day or it changes something.

What else can I try ?
Thanks
GLLM

ldrahnik, 2025-03-24:
@GLLM It works (is changeable) on Windows using official driver / app?

GLLM, 2025-03-25:
@ldrahnik I checked in Win11, on Asus apps (dialpad, proart hub & co) and could not alter the lightness of the led.
So maybe i'm chasing a three-legged duck ! :-D

ldrahnik, 2025-03-26:
@GLLM What we can do is only check how brightness reacts to writing all possible values to the register. But it still relies on hardware, it case it is not supported by hardware.. maybe asking official Asus support..  but if this option was not found in the official app and touchpad does not have icon on the top left corner for changing brightness..  so it would be supported but officially not.. non-technical support responding the questions will not know this..

ldrahnik, 2025-04-11:
Closed as probably not possible. Can be reopened.

closed 2025-04-11 by ldrahnik