asus-linux-drivers/asus-dialpad-driver #10: [Bug]: Nixos service cannot create virtual device

SamueleFacenda, 2025-04-07:
### Describe the bug

After moving the nixos service to an user service, the process doesn't have sufficent permissions to create a virtual device. I suspect that it doesn't have neither permissions to write to i2c. I tried writing a system service that runs with DynamicUser=true but there is impossible to grant access to the `XDG_RUNTIME_DIR` to connect to hyprland display (and maybe dbus?). 

### Expected behavior

The virtual device is created.

### Relevant log output

```shell
apr 07 11:24:57 zenbook dialpad.py[307318]: 2025-04-07 11:24:57,947 ERROR Error initializing virtual device: [Errno 13] Permission denied
apr 07 11:24:57 zenbook dialpad.py[307318]: 2025-04-07 11:24:57,947 ERROR Listening touchpad events unexpectedly failed
apr 07 11:24:57 zenbook dialpad.py[307318]: Traceback (most recent call last):
apr 07 11:24:57 zenbook dialpad.py[307318]:   File "/nix/store/ah95d2sc9s71fsmdd1jdblflgn4qqp49-python3.11-asus-dialpad-driver-0.0.1/share/asus-dialpad-driver/dialpad.py", line 375, in initialize_virtual_device
apr 07 11:24:57 zenbook dialpad.py[307318]:     uinput_device = dev.create_uinput_device()
apr 07 11:24:57 zenbook dialpad.py[307318]:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^
apr 07 11:24:57 zenbook dialpad.py[307318]:   File "/nix/store/w52gm2gwh22x2q62c4m854n8i64n3gpf-python3-3.11.11-env/lib/python3.11/site-packages/libevdev/device.py", line 764, in create_uinput_device
apr 07 11:24:57 zenbook dialpad.py[307318]:     d._uinput = UinputDevice(self._libevdev, uinput_fd)
apr 07 11:24:57 zenbook dialpad.py[307318]:                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
apr 07 11:24:57 zenbook dialpad.py[307318]:   File "/nix/store/w52gm2gwh22x2q62c4m854n8i64n3gpf-python3-3.11.11-env/lib/python3.11/site-packages/libevdev/_clib.py", line 964, in __init__
apr 07 11:24:57 zenbook dialpad.py[307318]:     raise OSError(-rc, os.strerror(-rc))
apr 07 11:24:57 zenbook dialpad.py[307318]: PermissionError: [Errno 13] Permission denied
apr 07 11:24:57 zenbook dialpad.py[307318]: During handling of the above exception, another exception occurred:
apr 07 11:24:57 zenbook dialpad.py[307318]: Traceback (most recent call last):
apr 07 11:24:57 zenbook dialpad.py[307318]:   File "/nix/store/ah95d2sc9s71fsmdd1jdblflgn4qqp49-python3.11-asus-dialpad-driver-0.0.1/share/asus-dialpad-driver/dialpad.py", line 1379, in <module>
apr 07 11:24:57 zenbook dialpad.py[307318]:     initialize_virtual_device()
apr 07 11:24:57 zenbook dialpad.py[307318]:   File "/nix/store/ah95d2sc9s71fsmdd1jdblflgn4qqp49-python3.11-asus-dialpad-driver-0.0.1/share/asus-dialpad-driver/dialpad.py", line 380, in initialize_virtual_device
apr 07 11:24:57 zenbook dialpad.py[307318]:     sys.exit(1)  # Exit if initialization fails
apr 07 11:24:57 zenbook dialpad.py[307318]:     ^^^^^^^^^^^
apr 07 11:24:57 zenbook dialpad.py[307318]: SystemExit: 1
apr 07 11:24:57 zenbook systemd[3292]: asus-dialpad-driver.service: Main process exited, code=exited, status=1/FAILURE
```

### Desktop

- Program version: 8b746614746fc8994ed8892759d6feb396035c17
- Distro: nixos unstable
- Kernel: 6.14.0

ldrahnik, 2025-04-08:
@SamueleFacenda Hello, could you please try the latest master? I have added config option user, by default should be used current user from the environment variable `$USER`.

ldrahnik, 2025-04-08:
@SamueleFacenda Reverted to be a non-user systemd service. Maybe would be better to have / to run under a specific user `asusdialpad`, can you test it (NixOS I personally do not use at all):

```
...
users.users.asusdialpad = {
  isSystemUser = true;
  extraGroups = [ "i2c" "input" "uinput" ];
};
...
serviceConfig = {
  User = "asusdialpad";
  Group = "asusdialpad";
  ...
}; 
```

Can be reopened.

SamueleFacenda, 2025-04-08:
Thanks, I will try but I don't think that it will be easy to do, the wayland display access is very strict.