asus-linux-drivers/asus-dialpad-driver #11: [Feature Request] proartp16 multi function

Ja7ad, 2025-04-19:
### Short Description

Hello

Please support multi function for proartp16, currently just do change volume.

ldrahnik, 2025-04-19:
@Ja7ad Hello, at first thank you for creating issue. 

What is the problem? Why do shortcuts not work (e.g. by default for firefox and code)? Name of opened window is not detected correctly? Your distribution info? Currently is supported getting window info via `KWin` for KDE with wayland, `org.gnome.Shell` for Gnome with Wayland, `_NET_WM_NAME` for x11.

```
app_shortcuts = {
    "code": {
        "center": [
          {"key": EV_KEY.KEY_MUTE, "trigger": "release", "duration": 1},
          {"key": EV_KEY.KEY_MUTE, "trigger": "release", "modifier": EV_KEY.KEY_LEFTSHIFT},
        ],
        "clockwise": [
          {"key": EV_KEY.KEY_VOLUMEUP, "trigger": "immediate"},
          {"key": EV_KEY.KEY_VOLUMEUP, "trigger": "immediate", "modifier": EV_KEY.KEY_LEFTSHIFT}
        ],
        "counterclockwise": [
          {"key": EV_KEY.KEY_VOLUMEDOWN, "trigger": "immediate"},
          {"key": EV_KEY.KEY_VOLUMEDOWN, "trigger": "immediate", "modifier": EV_KEY.KEY_LEFTSHIFT}
        ]
    },
    "firefox": {
        "center": [
          {"key": EV_KEY.KEY_MUTE, "trigger": "release"},
          {"key": EV_KEY.KEY_MUTE, "trigger": "release", "modifier": EV_KEY.KEY_LEFTSHIFT}
        ],
        "clockwise": [
          {"key": EV_KEY.KEY_VOLUMEUP, "trigger": "immediate"},
          {"key": EV_KEY.KEY_VOLUMEUP, "trigger": "immediate", "modifier": EV_KEY.KEY_LEFTSHIFT}
        ],
        "counterclockwise": [
          {"key": EV_KEY.KEY_VOLUMEDOWN, "trigger": "immediate"},
          {"key": EV_KEY.KEY_VOLUMEDOWN, "trigger": "immediate", "modifier": EV_KEY.KEY_LEFTSHIFT}
        ]
    },
    "none": {
        "center": [
          {"key": EV_KEY.KEY_MUTE, "trigger": "release"},
          {"key": EV_KEY.KEY_MUTE, "trigger": "release", "modifier": EV_KEY.KEY_LEFTSHIFT}
        ],
        "clockwise": [
          {"key": EV_KEY.KEY_VOLUMEUP, "trigger": "immediate"},
          {"key": EV_KEY.KEY_VOLUMEUP, "trigger": "immediate", "modifier": EV_KEY.KEY_LEFTSHIFT}
        ],
        "counterclockwise": [
          {"key": EV_KEY.KEY_VOLUMEDOWN, "trigger": "immediate"},
          {"key": EV_KEY.KEY_VOLUMEDOWN, "trigger": "immediate", "modifier": EV_KEY.KEY_LEFTSHIFT}
        ]
    }
}
```

Ja7ad, 2025-04-19:
> _NET_WM_NAME

Where is app_shortcut configure?

ldrahnik, 2025-04-19:
@Ja7ad `$ git clone` repository, change the appropriate layout in folder `layouts` to fit own needs and install, during install is then file copyied to `/usr/share/asus-dialpad-driver/layouts` and used for running.

Ja7ad, 2025-04-19:
Thank you @ldrahnik

Ja7ad, 2025-04-19:
For apply layout need to restart?

ldrahnik, 2025-04-19:
@Ja7ad Closing, create another issue if you face anything, have idea to improve etc., still this is only prototype, not even temporary driver. I can test driver aswell except position of circle for example because I do not own laptop from this series personally.

closed 2025-04-19 by ldrahnik