asus-linux-drivers/asus-dialpad-driver #13: [Bug]: Device not recognized

qippur, 2025-05-19:
### Describe the bug

As requested I am opening an issue for an unrecognized device:

### Expected behavior

the script should sow the name of my device (the large one in the pictures list)

### Relevant log output

```shell
$ bash install_device_check.sh
No i2c interface can be found. Make sure you have installed libevdev packages
$ dpkg -l | grep libevdev
ii  libevdev2:amd64                                          1.13.4+dfsg-1                       amd64        wrapper library for evdev devices
roberto@daitarn:~/devel/external/asus-dialpad-driver$
```

### Desktop

- Program version 0.0.1
- Distro Devuan Daedalus
- Kernel [e.g. 5.18.15.arch1-2]
- device: the large one in the pictures list

```
System Information
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: ProArt P16 H7606WI_H7606WI
Version: 1.0
Serial Number: redacted
UUID: redacted
Wake-up Type: Power Switch
SKU Number:
Family: ProArt P16
```

```
# lshw
daitarn
description: Notebook
product: ProArt P16 H7606WI_H7606WI
vendor: ASUSTeK COMPUTER INC.
version: 1.0
serial: redacted
width: 64 bits
capabilities: smbios-3.7.0 dmi-3.7.0 smp vsyscall32
configuration: boot=normal chassis=notebook family=ProArt P16 uuid=redacted
*-core
description: Motherboard
product: H7606WI
vendor: ASUSTeK COMPUTER INC.
physical id: 0
version: 1.0
serial: redacted
slot: Default string
*-firmware
description: BIOS
vendor: American Megatrends International, LLC.
physical id: 0
version: H7606WI.316
date: 02/11/2025
size: 64KiB
capacity: 32MiB
capabilities: pci upgrade shadowing cdboot bootselect socketedrom edd acpi biosbootspecification uefi
*-cache:0
description: L1 cache
physical id: b
slot: L1 - Cache
size: 960KiB
[... 751 lines cut ...]
*-input:10
product: WDHT1F01:00 2575:0926 Stylus
physical id: c
logical name: input8
logical name: /dev/input/event2
logical name: /dev/input/mouse1
capabilities: i2c
*-input:11
product: ASCF1A01:00 2808:0234 Mouse
physical id: d
logical name: input9
logical name: /dev/input/event3
logical name: /dev/input/mouse2
capabilities: i2c
*-network
description: Ethernet interface
physical id: e
bus info: usb@8:1
logical name: eth1
serial: redacted
size: 1Gbit/s
capacity: 1Gbit/s
capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
configuration: autonegotiation=on broadcast=yes driver=r8152 driverversion=v1.12.13 duplex=full firmware=rtl8153c-1 v2 08/19/21 ip=10.0.0.23 link=yes multicast=yes port=MII speed=1Gbit/s
```

ldrahnik, 2025-05-19:
@qippur Hello, thank you for creating an issue. Could you provide the output of this `$ sudo /usr/sbin/i2cdetect -l`

qippur, 2025-05-19:
Sure. Here you are

sudo /usr/sbin/i2cdetect -l
[sudo] password for :
i2c-0   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-2   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-3   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-4   smbus           SMBus PIIX4 adapter port 0 at 0b00      SMBus
adapter
i2c-5   smbus           SMBus PIIX4 adapter port 2 at 0b00      SMBus
adapter
i2c-6   smbus           SMBus PIIX4 adapter port 1 at 0b20      SMBus
adapter
i2c-7   i2c             AMDGPU DM i2c hw bus 0                  I2C adapter
i2c-8   i2c             AMDGPU DM i2c hw bus 1                  I2C adapter
i2c-9   i2c             AMDGPU DM i2c hw bus 2                  I2C adapter
i2c-10  i2c             AMDGPU DM i2c hw bus 3                  I2C adapter
i2c-11  i2c             AMDGPU DM i2c hw bus 4                  I2C adapter
i2c-12  i2c             AMDGPU DM i2c hw bus 5                  I2C adapter
i2c-13  i2c             AMDGPU DM i2c hw bus 6                  I2C adapter
i2c-14  i2c             AMDGPU DM i2c hw bus 7                  I2C adapter
i2c-15  i2c             AMDGPU DM i2c hw bus 8                  I2C adapter
i2c-16  i2c             AMDGPU DM aux hw bus 0                  I2C adapter
i2c-17  i2c             AMDGPU DM aux hw bus 1                  I2C adapter
i2c-18  i2c             AMDGPU DM aux hw bus 2                  I2C adapter
i2c-19  i2c             AMDGPU DM aux hw bus 3                  I2C adapter
i2c-20  i2c             AMDGPU DM aux hw bus 4                  I2C adapter
i2c-21  i2c             AMDGPU DM aux hw bus 5                  I2C adapter
i2c-22  i2c             AMDGPU DM aux hw bus 6                  I2C adapter
i2c-23  i2c             AMDGPU DM aux hw bus 7                  I2C adapter
i2c-24  i2c             AMDGPU DM aux hw bus 8                  I2C adapter
i2c-25  i2c             NVIDIA i2c adapter 1 at 64:00.0         I2C adapter
i2c-26  i2c             NVIDIA i2c adapter 3 at 64:00.0         I2C adapter
i2c-27  i2c             NVIDIA i2c adapter 5 at 64:00.0         I2C adapter

ldrahnik, 2025-05-19:
@qippur Now run in terminal command `$ sudo /usr/sbin/i2cdetect -l | grep DesignWare | sed -r "s/^(i2c\-[0-9]+).*/\1/"`

qippur, 2025-05-20:
Here you are

$ sudo /usr/sbin/i2cdetect -l | grep DesignWare | sed -r
"s/^(i2c\-[0-9]+).*/\1/"
[sudo] password:
i2c-0
i2c-1
i2c-2
i2c-

qippur, 2025-05-20:
> @qippur <https://github.com/qippur> Please, try the latest master branch.

Hi!

This is the output now.

$ bash ./install_device_check.sh
Testing interface i2c-0: success (adr 0x15)

For the sake of completeness, yesterday I tried to run the complete install
anyway.
Devuan ditches systemd.
The script made some breaking changes to my system.
It removed pam-elogind. I wasn't able to run wayland and pipewire as
XDG_RUNTIME_DIR wasn't set anymore.
I had to work quite a bit to find out what happened and to fix it.

The "How to start DialPad without systemd service?" section of README
doesn't cover this aspect.

ldrahnik, 2025-05-20:
@qippur What is the output when you try start `python3` script directly with no systemd service `of `/usr/share/asus-dialpad-driver/.env/bin/python3 /usr/share/asus-dialpad-driver/dialpad.py <up5401ea|e210ma|..>`)?

ldrahnik, 2025-05-20:
@qippur You can create a new environment with the required libraries in any directory like this:

```
$ virtualenv --python=$(command -v python3) ./.env
$ source ./.env/bin/activate
$ pip install --upgrade setuptools
$ pip install --upgrade pip
$ pip install -r requirements.txt
```

then:

```
<your-path>/.env/bin/python3 /usr/share/asus-dialpad-driver/dialpad.py <up5401ea|e210ma|..>
```

qippur, 2025-05-20:
The requirements.txt file includes systemd-python.

Should I remove it?

qippur, 2025-05-20:
I used pipenv and installed the virtual environment without systemd-python.

$ python3 dialpad.py
2025-05-20 11:08:15,238 INFO Wayland session detected and connected.
2025-05-20 11:08:15,238 ERROR DialPad layout *.py from dir layouts is
required as first argument. Re-run install script or add missing first
argument (valid value is default).

I don't know what argument to pass to the program. How can I identify
whether it is up5401ea or e210ma or another?

ldrahnik, 2025-05-20:
@qippur Use any string from the Readme.md block [layouts](https://github.com/asus-linux-drivers/asus-dialpad-driver#layouts)

qippur, 2025-05-20:
Here's the output

$ python3 dialpad.py proartp16
2025-05-20 11:45:33,055 INFO Wayland session detected and connected.
2025-05-20 11:45:33,056 INFO Detecting touchpad from string: "N:
Name="ASCF1A01:00 2808:0234 Touchpad""
2025-05-20 11:45:33,056 INFO Set touchpad device id 0 from S:
Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASCF1A01:00/0018:2808:0234.0003/input/input10
2025-05-20 11:45:33,056 INFO Set touchpad id 4 from H: Handlers=mouse3
event4
2025-05-20 11:45:33,056 INFO Detecting keyboard from string: "N: Name="Asus
Keyboard""
2025-05-20 11:45:33,056 INFO Set keyboard id 6 from H: Handlers=sysrq kbd
leds mouse5 rfkill event6
2025-05-20 11:45:33,056 ERROR Can't open the I2C bus connection (id: 0)

ldrahnik, 2025-05-20:
@qippur It looks like you have to set up i2c group, run `bash install_user_groups.sh`.

ldrahnik, 2025-05-20:
@qippur Did you rebooted?

qippur, 2025-05-20:
Well... this is not Windows, so no. Should I reboot?

ldrahnik, 2025-05-20:
@qippur Look at permissions of `i2c-dev` virtual directories, `$ ls -la`, maybe you do need reboot.

qippur, 2025-05-20:
After reboot it kind of works.

I can spot some errors connected with wayland and KDE.

$ python3 dialpad.py proartp16
2025-05-20 15:17:42,112 INFO Wayland session detected and connected.
2025-05-20 15:17:42,113 INFO Detecting touchpad from string: "N:
Name="ASCF1A01:00 2808:0234 Touchpad""
2025-05-20 15:17:42,113 INFO Set touchpad device id 0 from S:
Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-ASCF1A01:00/0018:2808:0234.0003/input/input10
2025-05-20 15:17:42,113 INFO Set touchpad id 4 from H: Handlers=mouse3
event4
2025-05-20 15:17:42,113 INFO Detecting keyboard from string: "N: Name="Asus
Keyboard""
2025-05-20 15:17:42,113 INFO Set keyboard id 6 from H: Handlers=sysrq kbd
leds mouse4 rfkill event6
2025-05-20 15:17:42,113 INFO Touchpad min-max: x 0-4762, y 0-3099
2025-05-20 15:17:42,114 INFO Virtual device initialized successfully.
2025-05-20 15:17:43,620 INFO Listening to keyboard events...
2025-05-20 15:17:43,621 INFO Listening to touchpad events...

***Touched top right button***

2025-05-20 15:17:48,733 INFO Top-right icon held for the required duration.
2025-05-20 15:17:48,734 INFO Setting up for config file key: "enabled" with
value: "True"
2025-05-20 15:17:48,834 INFO Toggling top-right icon: Enabling
WARNING: running xinput against an Xwayland server. See the xinput man page
for details.
unable to find device ASCF1A01:00 2808:0234 Touchpad
WARNING: running xinput against an Xwayland server. See the xinput man page
for details.
unable to find device ASCF1A01:00 2808:0234 Touchpad
WARNING: running xinput against an Xwayland server. See the xinput man page
for details.

***swoop around top left white led***

unable to find device ASCF1A01:00 2808:0234 Touchpad
2025-05-20 15:17:56,412 ERROR Error getting active KDE window title: [Errno
2] No such file or directory: 'qdbus'
2025-05-20 15:17:56,412 ERROR Error getting active GNOME Wayland window
title: name 'dbus' is not defined
2025-05-20 15:17:56,412 ERROR Unsupported session type or display not
connected.
2025-05-20 15:17:56,413 INFO Sent key press event: KEY_VOLUMEDOWN
2025-05-20 15:17:56,413 INFO Sent key release event: KEY_VOLUMEDOWN
2025-05-20 15:17:56,413 INFO Executed shortcut: KEY_VOLUMEDOWN with
modifier None (Held for 0.00s)
WARNING: running xinput against an Xwayland server. See the xinput man page
for details.
unable to find device ASCF1A01:00 2808:0234 Touchpad
WARNING: running xinput against an Xwayland server. See the xinput man page
for details.

***touched white led***

unable to find device ASCF1A01:00 2808:0234 Touchpad
2025-05-20 15:18:01,187 ERROR Error getting active KDE window title: [Errno
2] No such file or directory: 'qdbus'
2025-05-20 15:18:01,188 ERROR Error getting active GNOME Wayland window
title: name 'dbus' is not defined
2025-05-20 15:18:01,188 ERROR Unsupported session type or display not
connected.
2025-05-20 15:18:01,188 INFO Executed shortcut: KEY_MUTE with modifier None
(Held for 0.00s)
2025-05-20 15:18:01,366 ERROR Error getting active KDE window title: [Errno
2] No such file or directory: 'qdbus'
2025-05-20 15:18:01,366 ERROR Error getting active GNOME Wayland window
title: name 'dbus' is not defined
2025-05-20 15:18:01,366 ERROR Unsupported session type or display not
connected.
2025-05-20 15:18:01,366 INFO Sent key press event: KEY_MUTE
2025-05-20 15:18:01,366 INFO Sent key release event: KEY_MUTE
2025-05-20 15:18:01,366 INFO Executed shortcut: KEY_MUTE with modifier None
(Held for 0.19s)
WARNING: running xinput against an Xwayland server. See the xinput man page
for details.
unable to find device ASCF1A01:00 2808:0234 Touchpad

***CTRL-C***

^C2025-05-20 15:18:06,341 ERROR Listening touchpad events unexpectedly
failed
Traceback (most recent call last):
File "/home/common/asus-dialpad-driver/dialpad.py", line 1434, in <module>
listen_touchpad_events()
~~~~~~~~~~~~~~~~~~~~~~^^
File "/home/common/asus-dialpad-driver/dialpad.py", line 693, in
listen_touchpad_events
for event in d_t.events():
~~~~~~~~~~^^
File
"/home/rresecco/.local/share/virtualenvs/asus-dialpad-driver-lgyhuh20/lib/python3.13/site-packages/libevdev/device.py",
line 542, in events
ev = self._libevdev.next_event(flags)
File
"/home/rresecco/.local/share/virtualenvs/asus-dialpad-driver-lgyhuh20/lib/python3.13/site-packages/libevdev/_clib.py",
line 889, in next_event
rc = self._next_event(self._ctx, flags, ctypes.byref(ev))
KeyboardInterrupt
2025-05-20 15:18:06,343 INFO Clean up started
2025-05-20 15:18:06,345 INFO Setting up for config file key: "enabled" with
value: "False"
2025-05-20 15:18:06,445 INFO DialPad deactivated
2025-05-20 15:18:06,446 INFO Clean up finished
2025-05-20 15:18:06,446 INFO Exiting

Il giorno mar 20 mag 2025 alle ore 14:50 Lukáš Drahník <
***@***.***> ha scritto:

> *ldrahnik* left a comment (asus-linux-drivers/asus-dialpad-driver#13)
> <https://github.com/asus-linux-drivers/asus-dialpad-driver/issues/13#issuecomment-2894287498>
>
> @qippur <https://github.com/qippur> Look at permissions of i2c-dev
> virtual directories, $ ls -la, maybe you do need reboot.
>
> —
> Reply to this email directly, view it on GitHub
> <https://github.com/asus-linux-drivers/asus-dialpad-driver/issues/13#issuecomment-2894287498>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/ACXMXUJVU6F7Y4DXFQQOGQL27MQI3AVCNFSM6AAAAAB5NRMLZGVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDQOJUGI4DONBZHA>
> .
> You are receiving this because you were mentioned.Message ID:
> ***@***.***>
>

ldrahnik, 2025-05-20:
@qippur I forgot it was designed that you can skip only using `systemd`, so you should run in the standard way `bash install.sh` and then answer `N` to `Do you want install systemctl service? [y/N`, it would tell you about the necessary reboot.

qippur, 2025-05-20:
Great.

Should those messages not to worry?
If not, how do you suggest to proceed to perform a final installation?

closed 2025-05-20 by ldrahnik

qippur, 2025-05-20:
> @qippur <https://github.com/qippur> Does driver works as expected? If no,
> please create another issue with more description/this log. I will look at
> that further. Install it using bash install.sh, then answer N to systemd
> installation/using and then start it from the install directory like was
> mentioned in Readme.md: /usr/share/asus-dialpad-driver/.env/bin/python3
> /usr/share/asus-dialpad-driver/dialpad.py <asusvivobook16x|proartp16|..>

ok.

Thank you for your great support.